eProsima/Micro-XRCE-DDS-Agent #332: Huge Micro ros agent cpu usage that cause connection problems with clients

kalaajiahmad, 2023-04-24:
**Describe the bug**
Hi,
I am encountering huge CPU usage of the Micro ros agent running from docker as follows:
`docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy udp4 -p 8888`
[image]

**To Reproduce**
Steps to reproduce the behaviour:
1. Clone repo 'micro_ros_arduino' Foxy Branch
2. Flash 8 ESPs with it (Client Key unique for each Esp)
3. Run Publisher example for 3 hours
4. CPU spike happens after around 3h
5. See CPU spike on htop

**Expected behaviour**
A normal usage of ~100% CPU usage max at all times.

**System information (please complete the following information):**
 - OS: Ubuntu 20.04
 - ROS 2 Foxy
 - Hardware: 8x ESP32 micro-controllers running agent and client version 2.3.0

what can be the possible reason behind such a huge CPU usage?
Observation from my tests: it happens exponentially were it starts to increase to 200 then 500 then 1000%
And after the agent reaches past 200%, the Esp32 loses connection with the agent and fails to publish any messages.
Any help is much appreciated!

pablogs9, 2023-04-25:
Could you detail which entities does the 8 ESP32 have?

kalaajiahmad, 2023-04-25:
Yes for sure @pablogs9 
Currently the entities **on each** Esp32 are as follows: (So this is multiplied by 8, is that a problem for the agent?)
1 Node
5 Publishers
3 Subscribers
1 Timer
where i make sure to add the appropriate entities to the executor as such:
```c++
// create executor
executor = rclc_executor_get_zero_initialized_executor();
RCCHECK(rclc_executor_init(&executor, &support.context, 4, &allocator));
Serial.println("Created Executor!");
``` 

I also make sure to allocate the necessary memory for my string types by using this:
```c++
info.ssid.data = (char *)malloc(ROS_MSG_ALLOC * sizeof(char));
info.ssid.size = 0;
info.ssid.capacity = ROS_MSG_ALLOC;
``` 
where my ROS_MSG_ALLOC is 200

This is the colcon.meta config file used as well:
```cmake
"microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-UCLIENT_PROFILE_STREAM_FRAMING=ON",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
```

kalaajiahmad, 2023-04-25:
Also @pablogs9 
These are the Logs coming in from the agent when i saw 1000% CPU usage
```shell
[1682345616.693615] info   | UDPv4AgentLinux.cp | init                    | running...             | port: 28000
[1682345616.693966] info   | Root.cpp           | set_verbose_level       | logger setup           | verbose_level: 4
[1682345617.594863] info   | Root.cpp           | create_client           | create                 | client_key: 0x80EFFC02, session_id: 0x81
[1682345617.595038] info   | SessionManager.hpp | establish_session       | session established    | client_key: 0x80EFFC02, address: 192.168.96.1:23771
[1682345617.595110] info   | Root.cpp           | create_client           | create                 | client_key: 0xC4DCAA8D, session_id: 0x81
[1682345617.595137] info   | SessionManager.hpp | establish_session       | session established    | client_key: 0xC4DCAA8D, address: 192.168.96.1:14219
[1682345617.624950] info   | ProxyClient.cpp    | create_participant      | participant created    | client_key: 0x80EFFC02, participant_id: 0x000(1)
[1682345617.627878] info   | ProxyClient.cpp    | create_participant      | participant created    | client_key: 0xC4DCAA8D, participant_id: 0x000(1)
[1682345617.627943] info   | Root.cpp           | create_client           | create                 | client_key: 0xE5648135, session_id: 0x81
[1682345617.627960] info   | SessionManager.hpp | establish_session       | session established    | client_key: 0xE5648135, address: 192.168.96.1:24294
[1682345617.627973] info   | Root.cpp           | create_client           | create                 | client_key: 0xA1CD2AF7, session_id: 0x81
[1682345617.627980] info   | SessionManager.hpp | establish_session       | session established    | client_key: 0xA1CD2AF7, address: 192.168.96.1:35280
[1682345617.627997] info   | Root.cpp           | create_client           | create                 | client_key: 0xAE083B9A, session_id: 0x81
[1682345617.628013] info   | SessionManager.hpp | establish_session       | session established    | client_key: 0xAE083B9A, address: 192.168.96.1:55746
[1682345617.628022] info   | Root.cpp           | create_client           | create                 | client_key: 0x5629DD05, session_id: 0x81
[1682345617.628028] info   | SessionManager.hpp | establish_session       | session established    | client_key: 0x5629DD05, address: 192.168.96.1:42952
[1682345617.633926] info   | ProxyClient.cpp    | create_topic            | topic created          | client_key: 0x80EFFC02, topic_id: 0x000(2), participant_id: 0x000(1)
[1682345617.639066] info   | ProxyClient.cpp    | create_participant      | participant created    | client_key: 0xE5648135, participant_id: 0x000(1)
[1682345617.642982] info   | ProxyClient.cpp    | create_participant      | participant created    | client_key: 0xA1CD2AF7, participant_id: 0x000(1)
[1682345617.645712] info   | ProxyClient.cpp    | create_participant      | participant created    | client_key: 0xAE083B9A, participant_id: 0x000(1)
[1682345617.649870] info   | ProxyClient.cpp    | create_participant      | participant created    | client_key: 0x5629DD05, participant_id: 0x000(1)
[1682345617.649948] info   | Root.cpp           | create_client           | create                 | client_key: 0x9D9A1A6A, session_id: 0x81
[1682345617.649969] info   | SessionManager.hpp | establish_session       | session established    | client_key: 0x9D9A1A6A, address: 192.168.96.1:37312
[1682345617.650064] info   | ProxyClient.cpp    | create_topic            | topic created          | client_key: 0xC4DCAA8D, topic_id: 0x000(2), participant_id: 0x000(1)
[1682345617.650200] info   | ProxyClient.cpp    | create_publisher        | publisher created      | client_key: 0x80EFFC02, publisher_id: 0x000(3), participant_id: 0x000(1)
[1682345617.650290] info   | ProxyClient.cpp    | create_topic            | topic created          | client_key: 0xE5648135, topic_id: 0x000(2), participant_id: 0x000(1)
[1682345617.650976] info   | ProxyClient.cpp    | create_topic            | topic created          | client_key: 0xA1CD2AF7, topic_id: 0x000(2), participant_id: 0x000(1)
[1682345617.655965] info   | ProxyClient.cpp    | create_topic            | topic created          | client_key: 0xAE083B9A, topic_id: 0x000(2), participant_id: 0x000(1)
[1682345617.694752] info   | ProxyClient.cpp    | create_participant      | participant created    | client_key: 0x9D9A1A6A, participant_id: 0x000(1)
[1682345617.694920] info   | ProxyClient.cpp    | create_publisher        | publisher created      | client_key: 0xC4DCAA8D, publisher_id: 0x000(3), participant_id: 0x000(1)
[1682345617.696012] info   | ProxyClient.cpp    | create_datawriter       | datawriter created     | client_key: 0x80EFFC02, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1682345617.696168] info   | ProxyClient.cpp    | create_publisher        | publisher created      | client_key: 0xA1CD2AF7, publisher_id: 0x000(3), participant_id: 0x000(1)
[1682345617.696252] info   | ProxyClient.cpp    | create_topic            | topic created          | client_key: 0x5629DD05, topic_id: 0x000(2), participant_id: 0x000(1)
[1682345617.696404] info   | ProxyClient.cpp    | create_publisher        | publisher created      | client_key: 0xE5648135, publisher_id: 0x000(3), participant_id: 0x000(1)
[... cut ...]
[1682347959.568395] info   | ProxyClient.cpp    | create_topic            | topic created          | client_key: 0xAE083B9A, topic_id: 0x005(2), participant_id: 0x000(1)
[1682347960.176265] info   | ProxyClient.cpp    | create_topic            | topic created          | client_key: 0xAE083B9A, topic_id: 0x006(2), participant_id: 0x000(1)
[1682347960.192613] info   | ProxyClient.cpp    | create_topic            | topic created          | client_key: 0xAE083B9A, topic_id: 0x007(2), participant_id: 0x000(1)

```