eProsima/Micro-XRCE-DDS-Agent #337: Sending DDS message over specific network interface

henrykotze, 2023-07-13:
- Hardware description: Jetson Nano
- OS: We are running the foxy docker image for Jetson  
- Installation type: cmake 
- Version or commit hash: v2.3.0 

We are running the Agent on the Jetson Nano communicating via a router to a Nuttx device (PX4). We can see that the Agent is receiving heartbeat messages from the Nuttx device, and receiving messages from the ROS2 nodes running on the Jetson. However it is sending the messages it receives from the ROS2 Nodes on the loopback interface and not the physical network interface. 

Is there a way to explicitly tell the Agent to use a specific network interface? 

Thanks

Acuadros95, 2023-07-13:
> Is there a way to explicitly tell the Agent to use a specific network interface?

To accomplish this, you can configure the transport of the Fast dds entities of the Agent. 
This can be done with a XML profile, take a look at this documentation: [Link](https://fast-dds.docs.eprosima.com/en/latest/fastdds/xml_configuration/transports.html).

If support on this part is needed, feel free to open an issue at FastDDS  repo.

Acuadros95, 2023-07-24:
Closing as a solution was provided

closed 2023-07-24 by Acuadros95

Keller253, 2026-04-05:
> [@Keller253](https://github.com/Keller253)
> 
> Yes we managed to resolved this issue. I cant remember how, but our current implementation is using the physical network interface.
> 
> Perhaps if you explain your setup, versions, i can assit

@henrykotze Thx for your reply. I have the following setup using Docker:

1. `Micro-XRCE-DDS Agent Container`: Executes the Micro XRCE DDS Agent using the official `ros-humble` image. It receives messages from a client and should publish them as ROS2 topics.
2. `Topic Listener Container`: Additional container that is based on the official `ros-humble` image and listens to the topics published by the `Micro-XRCE-DDS Agent Container`.

This setup seems to have the same issue as you had: Due to the Agent publishing the ROS2 topics and messages only on the loopback interface the messages are not received by the additional `Topic Listener Container`. If I execute `ROS_LOCALHOST=1 && ros2 topic echo my/topic` within the `Micro-XRCE-DDS Agent Container` I can view the incoming messages that are created by the agent.

I hope I can solve this problem by configuring the agent to use the external interface, which will make the messages visible to the other container as well. That’s why I’m very interested in how you managed to use the other network interface.

Additional remark: All containers use `FASTDDS_BUILTIN_TRANSPORTS=UDPv4`.