eProsima/Micro-XRCE-DDS-Agent #340: Baud rate is not set in SerialAgent on macOS

srmainwaring, 2023-08-17:
## Overview

The Micro-XRCE-DDS-Agent does not support a hardware serial transport on macOS because the baudrate is not set correctly due to the linux specific conditional compilation flags:

- `_HAVE_STRUCT_TERMIOS_C_ISPEED`
- `_HAVE_STRUCT_TERMIOS_C_OSPEED`

which are set in:

https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/40954c2379dfcb2bfa9f6ba22146c57c6742b5b7/include/uxr/agent/utils/ArgumentParser.hpp#L1071-L1076

https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/40954c2379dfcb2bfa9f6ba22146c57c6742b5b7/src/cpp/transport/serial/MultiTermiosAgentLinux.cpp#L88-L93

https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/40954c2379dfcb2bfa9f6ba22146c57c6742b5b7/src/cpp/transport/serial/TermiosAgentLinux.cpp#L105-L110

The macOS implementation of `termios.h` does not define `_HAVE_STRUCT_TERMIOS_C_ISPEED` etc. One fix would be to modify the `#if #endif` check to:

```c++
#if _HAVE_STRUCT_TERMIOS_C_ISPEED || __APPLE__
            cfsetispeed(&new_attrs, termios_attrs_.c_ispeed);
#endif
#if _HAVE_STRUCT_TERMIOS_C_OSPEED || __APPLE__
            cfsetospeed(&new_attrs, termios_attrs_.c_ospeed);
#endif
```

## Issue template

- Hardware description: MacPro 2019 and MacBook Pro M1
- OS: macOS Ventura 13.5
- Installation type: build from source
- Version or commit hash: 
    - remote: https://github.com/srmainwaring/micro-ROS-Agent.git
    - branch: humble-macos
    - hash: 64001f06e0075a727eaaac2e24d0b156f43df344

- Compile flags:

    Unlink `spdlog` as the `brew` version conflicts:

    ```bash
    brew unlink spdlog
    ```

    Build with logging (do not use system logger).

    ```bash
    colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo - 
    DCMAKE_MACOSX_RPATH=FALSE -DUAGENT_SOCKETCAN_PROFILE=OFF -DUAGENT_LOGGER_PROFILE=ON  - 
    DUAGENT_USE_SYSTEM_LOGGER=OFF -DUAGENT_USE_SYSTEM_FASTDDS=ON - 
    DUAGENT_USE_SYSTEM_FASTCDR=ON --packages-select micro_ros_agent --cmake-force
    ```

#### Steps to reproduce the issue

- Install the `int32_publisher_custom_transport` from https://github.com/micro-ROS/micro_ros_espidf_component on an esp32.
- Run the micro-ROS agent from the project workspace with verbose logging enabled:

    ```bash
    ./build/micro_ros_agent/micro_ros_agent serial -D /dev/cu.usbserial-0001 -b 115200 -v6
    [1692281492.147983] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
    [1692281492.148393] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
    ```

#### Expected behavior

- The client connection is established.
- The topics are visible:
    ```
    $ ros2 topic list                          
    /freertos_int32_publisher
    /parameter_events
    /rosout
    ```

#### Actual behavior

- The client does not connect
- The topics and node are not visible 

#### Additional information

- Running on a slightly customised version of the `humble` branch.
- The test works on an Ubuntu 22.04 VM hosted on macOS.

pablogs9, 2023-08-18:
Hello Micro XRCE-DDS Agent does not provide support for macOS: https://micro-xrce-dds.docs.eprosima.com/en/latest/agent.html

If this is a small fix, please open a PR with your required changes, but we do not have plans for maintaining macOS platform.

srmainwaring, 2023-08-18:
> If this is a small fix, please open a PR with your required changes

Will do, it's a small change to the three files listed above.

> but we do not have plans for maintaining macOS platform.

It's not a great deal of work to run the entire eProsima suite on macOS (micro-ROS agent, integration services, fast-dds spy etc). Mostly small fixes here and there such as https://github.com/eProsima/Integration-Service/pull/196. I have working branches with the required changes which I'd be happy to contribute.

pablogs9, 2023-08-18:
> which I'd be happy to contribute.

That would be nice thanks!

srmainwaring, 2023-09-07:
> That would be nice thanks!

I've posted PRs for fixes to support Fast DDS spy and Fast DDS monitor on macOS. It's very useful to have the tools available natively on macs as docker networking is complicated to set up on macOS (because there is an extra VM involved).

PR #342 (2023-09-08): Enable termios baudrate configuration for macOS
Replace https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/341 to trigger CI. See: https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/341#issuecomment-1711174707.

Closes #340

Ensure that the baud rate is set in the termios attributes when compiling with `clang` on macOS.