eProsima/Micro-XRCE-DDS-Agent #344: Cannot build as a ROS2 package because of reference to outdated spdlog

slim71, 2023-09-23:
## Issue template

- Hardware description: Lenovo Yoga 720
- OS: Ubuntu 22.04
- Installation type: ROS2 package (using `colcon build`)
- Version or commit hash: tag **v2.4.0** and newer

#### Steps to reproduce the issue
Just follow the installation instructions to [build within a ROS2 workspace](https://docs.px4.io/main/en/middleware/uxrce_dds.html#build-run-within-ros-2-workspace), then try to build with `colcon build`.

#### Expected behavior
The project should build successfully with no issues

#### Actual behavior
The project does not build, returning errors related to the `fmt` library, linked through `spdlog`:

```shell
$ colcon build 
Starting >>> microxrcedds_agent
--- stderr: microxrcedds_agent                             
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  /opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:51 (find_package)
  CMakeLists.txt:153 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

In file included from /usr/include/spdlog/spdlog.h:12,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/logger/Logger.hpp:23,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/client/session/stream/OutputStream.hpp:22,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/client/session/Session.hpp:20,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/client/ProxyClient.hpp:20,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/src/cpp/publisher/Publisher.cpp:16:
/usr/include/spdlog/common.h:127:111: error: ‘basic_runtime’ is not a member of ‘fmt’
  127 |           std::is_convertible<T, fmt::basic_string_view<Char>>::value || std::is_same<remove_cvref_t<T>, fmt::basic_runtime<Char>>::value>
      |                                                                                                               ^~~~~~~~~~~~~
/usr/include/spdlog/common.h:127:111: error: ‘basic_runtime’ is not a member of ‘fmt’
/usr/include/spdlog/common.h:127:125: error: template argument 2 is invalid
  127 |           std::is_convertible<T, fmt::basic_string_view<Char>>::value || std::is_same<remove_cvref_t<T>, fmt::basic_runtime<Char>>::value>
      |                                                                                                                             ^~~~
/usr/include/spdlog/common.h:127:138: error: expected ‘{’ before ‘>’ token
  127 |           std::is_convertible<T, fmt::basic_string_view<Char>>::value || std::is_same<remove_cvref_t<T>, fmt::basic_runtime<Char>>::value>
      |                                                                                                                                          ^
gmake[5]: *** [CMakeFiles/microxrcedds_agent.dir/build.make:174: CMakeFiles/microxrcedds_agent.dir/src/cpp/publisher/Publisher.cpp.o] Error 1
gmake[5]: *** Waiting for unfinished jobs....
In file included from /usr/include/spdlog/spdlog.h:12,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/logger/Logger.hpp:23,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/client/session/stream/OutputStream.hpp:22,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/client/session/Session.hpp:20,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/client/ProxyClient.hpp:20,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/src/cpp/topic/Topic.cpp:17:
/usr/include/spdlog/common.h:127:111: error: ‘basic_runtime’ is not a member of ‘fmt’
  127 |           std::is_convertible<T, fmt::basic_string_view<Char>>::value || std::is_same<remove_cvref_t<T>, fmt::basic_runtime<Char>>::value>
      |                                                                                                               ^~~~~~~~~~~~~
/usr/include/spdlog/common.h:127:111: error: ‘basic_runtime’ is not a member of ‘fmt’
/usr/include/spdlog/common.h:127:125: error: template argument 2 is invalid
  127 |           std::is_convertible<T, fmt::basic_string_view<Char>>::value || std::is_same<remove_cvref_t<T>, fmt::basic_runtime<Char>>::value>
      |                                                                                                                             ^~~~
/usr/include/spdlog/common.h:127:138: error: expected ‘{’ before ‘>’ token
  127 |           std::is_convertible<T, fmt::basic_string_view<Char>>::value || std::is_same<remove_cvref_t<T>, fmt::basic_runtime<Char>>::value>
      |                                                                                                                                          ^
gmake[5]: *** [CMakeFiles/microxrcedds_agent.dir/build.make:160: CMakeFiles/microxrcedds_agent.dir/src/cpp/topic/Topic.cpp.o] Error 1
In file included from /usr/include/spdlog/spdlog.h:12,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/logger/Logger.hpp:23,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/client/session/stream/OutputStream.hpp:22,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/client/session/Session.hpp:20,
[... 222 lines cut ...]
      |                                                                                                                             ^~~~
/usr/include/spdlog/common.h:127:138: error: expected ‘{’ before ‘>’ token
  127 |           std::is_convertible<T, fmt::basic_string_view<Char>>::value || std::is_same<remove_cvref_t<T>, fmt::basic_runtime<Char>>::value>
      |                                                                                                                                          ^
gmake[5]: *** [CMakeFiles/microxrcedds_agent.dir/build.make:90: CMakeFiles/microxrcedds_agent.dir/src/cpp/AgentInstance.cpp.o] Error 1
gmake[4]: *** [CMakeFiles/Makefile2:85: CMakeFiles/microxrcedds_agent.dir/all] Error 2
gmake[3]: *** [Makefile:136: all] Error 2
gmake[2]: *** [CMakeFiles/uagent.dir/build.make:86: uagent-prefix/src/uagent-stamp/uagent-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/uagent.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
---
Failed   <<< microxrcedds_agent [7.07s, exited with code 2]

Summary: 0 packages finished [7.28s]
  1 package failed: microxrcedds_agent
  1 package had stderr output: microxrcedds_agent
```

#### Additional information
Before opening this issue, I've tried to search on the internet.
I've found the same problem mentioned inside the `spdlog` repository, issue [spdlog::#2825](https://github.com/gabime/spdlog/issues/2825).
Since `spdlog` is linked through the *CMakelists.txt* file, I could not just change the flag mentioned there as I would have done if I was building it from source.
Following the other issues linked there, namely [spdlog::#2805](https://github.com/gabime/spdlog/issues/2805), [fmt::#3318](https://github.com/fmtlib/fmt/issues/3318) and [fmt::#2919](https://github.com/fmtlib/fmt/issues/2919) and inspecting the `spdlog` repo, I've seen that this issue has been already resolved in newer versions of `spdlog`. Changing the one referenced in the *CMakelists.txt* file should be enough to close this.

**Note**: this might occur even when building as a standalone project, but I haven't tried that since I need it as a ROS2 package.

pablogs9, 2023-09-25:
Which distro of ROS 2 you have installed?

Have you tried to build against your local SPDLog installation using the flag `UAGENT_USE_SYSTEM_LOGGER`?
https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/40954c2379dfcb2bfa9f6ba22146c57c6742b5b7/CMakeLists.txt#L32

slim71, 2023-09-25:
I've got Humble.

Did not try to use that flag, since I do not have any local SDPLog. I might also have missed that flag and any info about the need for a local one. Maybe this issue was already known then?

pablogs9, 2023-09-26:
Yes, I have seen this issue in the past: https://github.com/micro-ROS/micro-ROS-Agent/issues/186 or https://github.com/micro-ROS/micro-ROS-Agent/issues/50

If I recall correctly Humble has an spdlog version installed, at least, the docker image has it:

[image]

Try to build with the flag `UAGENT_USE_SYSTEM_LOGGER` enabled.

slim71, 2023-09-26:
I tried but did not manage to make it work.
I've tried setting it to `ON` from the source, and also tried from CLI in multiple ways to avoid errors (note that I have to select the package since I have many in the workspace):
 - `colcon build --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON --packages-select microxrcedds_agent`
 - `colcon build --cmake-args " -DUAGENT_USE_SYSTEM_LOGGER=ON" --packages-select microxrcedds_agent`
 - `colcon build --packages-select microxrcedds_agent --cmake-args " -DUAGENT_USE_SYSTEM_LOGGER=ON"`
 - `colcon build --packages-select=microxrcedds_agent --cmake-args " -DUAGENT_USE_SYSTEM_LOGGER=ON"`
 - `colcon build --packages-select microxrcedds_agent --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON`
 - `colcon build --packages-select=microxrcedds_agent --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON`
but nothing worked.
For completeness, here's the search results I have with what you've shown
[image]

Am I missing something here?

pablogs9, 2023-09-27:
Did you start in a clean environment? Ensure that build, log and install folder are removed before running the colcon command

slim71, 2023-09-27:
I did, and even retried now: no luck
```shell
slim71@slim71-Ubuntu:~/Documents/git/SpartanLIFT$ ls -l build
total 0
slim71@slim71-Ubuntu:~/Documents/git/SpartanLIFT$ ls -l install/
total 0
slim71@slim71-Ubuntu:~/Documents/git/SpartanLIFT$ ls -l log/
total 0
slim71@slim71-Ubuntu:~/Documents/git/SpartanLIFT$ colcon build --packages-select microxrcedds_agent --cmake-args " -DUAGENT_USE_SYSTEM_LOGGER=ON"
[0.368s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/slim71/Documents/git/SpartanLIFT/install/reach' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.368s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/slim71/Documents/git/SpartanLIFT/install/px4_ros_com' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.368s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/slim71/Documents/git/SpartanLIFT/install/pelican' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.368s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/slim71/Documents/git/SpartanLIFT/install/px4_msgs' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.369s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/slim71/Documents/git/SpartanLIFT/install/px4' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.369s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/slim71/Documents/git/SpartanLIFT/install/odst' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.369s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/slim71/Documents/git/SpartanLIFT/install/comms' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.369s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/slim71/Documents/git/SpartanLIFT/install/cargo' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.370s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/slim71/Documents/git/SpartanLIFT/install/reach' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.370s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/slim71/Documents/git/SpartanLIFT/install/px4_ros_com' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.370s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/slim71/Documents/git/SpartanLIFT/install/pelican' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.370s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/slim71/Documents/git/SpartanLIFT/install/px4_msgs' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.370s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/slim71/Documents/git/SpartanLIFT/install/px4' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.370s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/slim71/Documents/git/SpartanLIFT/install/comms' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.370s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/slim71/Documents/git/SpartanLIFT/install/cargo' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> microxrcedds_agent
--- stderr: microxrcedds_agent                             
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  /opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:51 (find_package)
  CMakeLists.txt:153 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

In file included from /usr/include/spdlog/spdlog.h:12,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/logger/Logger.hpp:23,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/client/session/stream/OutputStream.hpp:22,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/client/session/Session.hpp:20,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/include/uxr/agent/client/ProxyClient.hpp:20,
                 from /home/slim71/Documents/git/SpartanLIFT/src/Micro-XRCE-DDS-Agent/src/cpp/participant/Participant.cpp:16:
/usr/include/spdlog/common.h:127:111: error: ‘basic_runtime’ is not a member of ‘fmt’
  127 |           std::is_convertible<T, fmt::basic_string_view<Char>>::value || std::is_same<remove_cvref_t<T>, fmt::basic_runtime<Char>>::value>
      |                                                                                                               ^~~~~~~~~~~~~
/usr/include/spdlog/common.h:127:111: error: ‘basic_runtime’ is not a member of ‘fmt’
/usr/include/spdlog/common.h:127:125: error: template argument 2 is invalid
  127 |           std::is_convertible<T, fmt::basic_string_view<Char>>::value || std::is_same<remove_cvref_t<T>, fmt::basic_runtime<Char>>::value>
      |                                                                                                                             ^~~~
/usr/include/spdlog/common.h:127:138: error: expected ‘{’ before ‘>’ token
[... cut ...]
gmake[4]: *** [CMakeFiles/Makefile2:85: CMakeFiles/microxrcedds_agent.dir/all] Error 2
gmake[3]: *** [Makefile:136: all] Error 2
gmake[2]: *** [CMakeFiles/uagent.dir/build.make:86: uagent-prefix/src/uagent-stamp/uagent-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/uagent.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
---
Failed   <<< microxrcedds_agent [5.76s, exited with code 2]

Summary: 0 packages finished [5.96s]
  1 package failed: microxrcedds_agent
  1 package had stderr output: microxrcedds_agent
```

pablogs9, 2023-09-27:
Are you sure that you have a clean humble installation?

I have just tried to replicate in a clean ROS 2 Humble docker with no problem:

https://asciinema.org/a/6aU9thzS4ivJTzBH0yD930EkQ

slim71, 2023-09-27:
Not clean aka new, since I'm working with it, but it should be all good and with no problems...
I'll check back once I get back to it, since I'm out now...
Any specific suggestions as what to look out for? Or additional info that might prove useful?

pablogs9, 2023-09-27:
Just check the differences between your environment and a clean one, and make sure that you are not installing/modifying spdlog anywhere.

slim71, 2023-09-28:
I've tried to check everything I could think of:
- executed `ros2 doctor`
- updated all ROS2 packages
- tried running the "standard"/demo package of `turtlesim` (+ `ros2 doctor` while running)
- checked for additional `spdlog` installations, but I could not find any (only some VSCode folders, in addition to the same ones you showed)
[image]
- checked status of `colcon`
- tried to take a look at all logs, with no luck

I can provide my logs if it can be useful, but other than that I think I'm out of options...

pablogs9, 2023-09-29:
Is it not possible for you to use the dockerized version of the micro-ROS Agent? https://hub.docker.com/r/microros/micro-ros-agent

slim71, 2023-09-29:
Unfortunately not for this... But it works, with the changes I've made in the PR linked to this ticket

Mendeler, 2023-12-05:
> Unfortunately not for this... But it works, with the changes I've made in the PR linked to this ticket

Sorry,` I'm not familiar with GitHub, I also encountered this compilation problem with Ubuntu22.04+humble, how did you solve it, thank you?