eProsima/Micro-XRCE-DDS-Agent #347: SIGABRT: "Not enough memory in the buffer stream" with rqt

yashi, 2023-09-28:
Hi,

Does anyone successfully run rqt with micro-ros Agent?

On my PC, the agent dies when I run both rqt and micro-ROS. That is, the Agent doesn't die when there are only:
- the agent and micro-ROS nodes on MCU
- the agent and rqt without plugins
- the agent and rqt with node graph plugin loaded

But as soon as I run the three, Agent, micro-ROS nodes, and rqt with the "Node Graph", Agent aborts with the following message:
```
terminate called after throwing an instance of 'eprosima::fastcdr::exception::NotEnoughMemoryException'
  what():  Not enough memory in the buffer stream
[ros2run]: Aborted
```

Full backtrace is down below.

## Issue template

- Hardware description: PC
- OS: Debian GNU/Linux Sid
- Installation type: my own build
- micro-ROS Agent version: 3.0.5
- rqt version: 1.3.3

#### Steps to reproduce the issue
1. Run micro-ROS Agent on a PC
2. Run rqt on the same PC
3. Run micro-ROS on a MCU
4. The agent dies with SIGABRT

#### Expected behavior
Agent doesn't die.

#### Actual behavior
Agent dies with SIGABRT.

#### Additional information

<details>
<summary>Backtrace</summary>
<detail>

```
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
#1  0x00007ffff5aa815f in __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x00007ffff5a5a472 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff5a444b2 in __GI_abort () at ./stdlib/abort.c:79
#4  0x00007ffff5ca09db in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff5caffaa in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff5cb0015 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff5cb0268 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff7d843ec in eprosima::fastcdr::Cdr::deserializeArray (this=0x7fffd6ffaaa0, char_t=0x7fff38002900 "", numElements=24)
    at /home/yashi/work/yoshida/humble/src/eProsima/Fast-CDR/src/cpp/Cdr.cpp:2076
#9  0x00007ffff7a5a7dd in eprosima::fastcdr::Cdr::deserializeArray (this=0x7fffd6ffaaa0, octet_t=0x7fff38002900 "", numElements=24)
    at /home/yashi/work/yoshida/humble/install/include/fastcdr/Cdr.h:2712
#10 0x00007ffff7d58185 in eprosima::fastcdr::Cdr::deserialize<unsigned char, 24ul> (this=0x7fffd6ffaaa0, array_t=...)
    at /home/yashi/work/yoshida/humble/install/include/fastcdr/Cdr.h:2528
#11 0x00007ffff7d580f9 in eprosima::fastcdr::Cdr::operator>><unsigned char, 24ul> (this=0x7fffd6ffaaa0, array_t=...)
    at /home/yashi/work/yoshida/humble/install/include/fastcdr/Cdr.h:767
#12 0x00007ffff7d57df6 in rmw_dds_common::msg::typesupport_fastrtps_cpp::cdr_deserialize (cdr=..., ros_message=...)
    at /home/yashi/work/yoshida/humble/build/rmw_dds_common/rosidl_typesupport_fastrtps_cpp/rmw_dds_common/msg/detail/dds_fastrtps/gid__type_support.cpp:50
#13 0x00007ffff7d58393 in rmw_dds_common::msg::typesupport_fastrtps_cpp::cdr_deserialize (cdr=..., ros_message=...)
    at /home/yashi/work/yoshida/humble/build/rmw_dds_common/rosidl_typesupport_fastrtps_cpp/rmw_dds_common/msg/detail/dds_fastrtps/node_entities_info__type_support.cpp:107
#14 0x00007ffff7d59ca8 in rmw_dds_common::msg::typesupport_fastrtps_cpp::cdr_deserialize (cdr=..., ros_message=...)
    at /home/yashi/work/yoshida/humble/build/rmw_dds_common/rosidl_typesupport_fastrtps_cpp/rmw_dds_common/msg/detail/dds_fastrtps/participant_entities_info__type_support.cpp:117
#15 0x00007ffff7d59f2d in rmw_dds_common::msg::typesupport_fastrtps_cpp::_ParticipantEntitiesInfo__cdr_deserialize (cdr=..., untyped_ros_message=0x7fffd6ffaf00)
    at /home/yashi/work/yoshida/humble/build/rmw_dds_common/rosidl_typesupport_fastrtps_cpp/rmw_dds_common/msg/detail/dds_fastrtps/participant_entities_info__type_support.cpp:234
[... 25 lines cut ...]
    at /home/yashi/work/yoshida/humble/src/eProsima/Fast-DDS/src/cpp/rtps/messages/MessageReceiver.cpp:1081
#29 0x00007ffff6a237aa in eprosima::fastrtps::rtps::MessageReceiver::findAllReaders<eprosima::fastrtps::rtps::MessageReceiver::proc_Submsg_Heartbeat(eprosima::fastrtps::rtps::CDRMessage_t*, eprosima::fastrtps::rtps::SubmessageHeader_t*) const::<lambda(eprosima::fastrtps::rtps::RTPSReader*)> >(const eprosima::fastrtps::rtps::EntityId_t &, const struct {...} &) const (this=0x7fffe80dfc40, readerID=..., callback=...) at /home/yashi/work/yoshida/humble/src/eProsima/Fast-DDS/src/cpp/rtps/messages/MessageReceiver.cpp:664
#30 0x00007ffff6a22586 in eprosima::fastrtps::rtps::MessageReceiver::proc_Submsg_Heartbeat (this=0x7fffe80dfc40, msg=0x7fffd6ffb960, smh=0x7fffd6ffb858)
    at /home/yashi/work/yoshida/humble/src/eProsima/Fast-DDS/src/cpp/rtps/messages/MessageReceiver.cpp:1078
#31 0x00007ffff6a1f826 in eprosima::fastrtps::rtps::MessageReceiver::processCDRMsg (this=0x7fffe80dfc40, source_locator=..., reception_locator=..., msg=0x7fffd6ffb960)
    at /home/yashi/work/yoshida/humble/src/eProsima/Fast-DDS/src/cpp/rtps/messages/MessageReceiver.cpp:479
#32 0x00007ffff6a346d3 in eprosima::fastrtps::rtps::ReceiverResource::OnDataReceived (this=0x7fffe80a8d50, data=0x7fffe01981b0 "RTPS\002\002\001\017\001\017\v<\t\030v'", size=52, 
    localLocator=..., remoteLocator=...) at /home/yashi/work/yoshida/humble/src/eProsima/Fast-DDS/src/cpp/rtps/network/ReceiverResource.cpp:121
#33 0x00007ffff6e633b7 in eprosima::fastdds::rtps::SharedMemChannelResource::perform_listen_operation (this=0x7fffe80a8110, input_locator=...)
    at /home/yashi/work/yoshida/humble/src/eProsima/Fast-DDS/src/cpp/rtps/transport/shared_mem/SharedMemChannelResource.hpp:146
#34 0x00007ffff6e7ad50 in std::__invoke_impl<void, void (eprosima::fastdds::rtps::SharedMemChannelResource::*)(eprosima::fastrtps::rtps::Locator_t), eprosima::fastdds::rtps::SharedMemChannelResource*, eprosima::fastrtps::rtps::Locator_t> (
    __f=@0x7fffe80dfa88: (void (eprosima::fastdds::rtps::SharedMemChannelResource::*)(eprosima::fastdds::rtps::SharedMemChannelResource * const, eprosima::fastrtps::rtps::Locator_t)) 0x7ffff6e6318e <eprosima::fastdds::rtps::SharedMemChannelResource::perform_listen_operation(eprosima::fastrtps::rtps::Locator_t)>, __t=@0x7fffe80dfa80: 0x7fffe80a8110)
    at /usr/include/c++/12/bits/invoke.h:74
#35 0x00007ffff6e7a946 in std::__invoke<void (eprosima::fastdds::rtps::SharedMemChannelResource::*)(eprosima::fastrtps::rtps::Locator_t), eprosima::fastdds::rtps::SharedMemChannelResource*, eprosima::fastrtps::rtps::Locator_t> (
    __fn=@0x7fffe80dfa88: (void (eprosima::fastdds::rtps::SharedMemChannelResource::*)(eprosima::fastdds::rtps::SharedMemChannelResource * const, eprosima::fastrtps::rtps::Locator_t)) 0x7ffff6e6318e <eprosima::fastdds::rtps::SharedMemChannelResource::perform_listen_operation(eprosima::fastrtps::rtps::Locator_t)>) at /usr/include/c++/12/bits/invoke.h:96
#36 0x00007ffff6e7a747 in std::thread::_Invoker<std::tuple<void (eprosima::fastdds::rtps::SharedMemChannelResource::*)(eprosima::fastrtps::rtps::Locator_t), eprosima::fastdds::rtps::SharedMemChannelResource*, eprosima::fastrtps::rtps::Locator_t> >::_M_invoke<0ul, 1ul, 2ul> (this=0x7fffe80dfa68) at /usr/include/c++/12/bits/std_thread.h:279
#37 0x00007ffff6e7a300 in std::thread::_Invoker<std::tuple<void (eprosima::fastdds::rtps::SharedMemChannelResource::*)(eprosima::fastrtps::rtps::Locator_t), eprosima::fastdds::rtps::SharedMemChannelResource*, eprosima::fastrtps::rtps::Locator_t> >::operator() (this=0x7fffe80dfa68) at /usr/include/c++/12/bits/std_thread.h:286
#38 0x00007ffff6e79ce8 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (eprosima::fastdds::rtps::SharedMemChannelResource::*)(eprosima::fastrtps::rtps::Locator_t), eprosima::fastdds::rtps::SharedMemChannelResource*, eprosima::fastrtps::rtps::Locator_t> > >::_M_run (this=0x7fffe80dfa60) at /usr/include/c++/12/bits/std_thread.h:231
#39 0x00007ffff5cdc433 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#40 0x00007ffff5aa63ec in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:444
#41 0x00007ffff5b26a2c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) 
```

</detail>

pablogs9, 2023-09-28:
Which distro of ROS 2 are you running?

yashi, 2023-09-28:
Humble.

pablogs9, 2023-09-29:
Could you confirm that using `iron` version of the agent this issue does not happens?

In my scenario, I can replicate your issue in `humble` but if I just run the `iron` agent, it does not happens.

If you confirm that, I will take a look at humble

yashi, 2023-09-29:
You are right.  My RQT was somehow built against `iron`. I apologize and thank you for your time.

| | micro-ROS | Agent | RQT |
|----|----|----|----|
| NG | Humble | Humble | v1.3.3 built against Iron |
| OK | Humble | Humble | v1.3.3 built against Humble |
| OK | Humble | Iron | v1.3.3 built against Iron |

closed 2023-09-29 by yashi