eProsima/Micro-XRCE-DDS-Agent #350: XRCE_DOMAIN_ID_OVERRIDE doesn't work. It is overwritten by dds_xrce_profile.xml

KyleJewiss, 2023-10-27:
## Issue template

- Hardware description: Lenovo Laptop. Cube Orange on a PX4 
- OS:  Ubuntu 22.04
- Installation type: FastDDS version
- Version or commit hash: 2.4.1

#### Steps to reproduce the issue
We are using ardupilot on a Cube Orange/PX4. We build and upload the rover version onto the cube orange, which all works very well. We then have integrated Micro-XRCE-DDS-Gen for ardupilot following the documentation here:
https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_DDS

We changed the standard domain_ID on the microROS client side from '0' to '255' to enable the ability to override the domain on the agent side. The changes we made can be seen here:
https://github.com/Greenroom-Robotics/ardupilot/commit/69143511f1c6b97315c5c9aad62743b0c8145ada

We can then run the following command to post the data from ardupilot to the desired ROS domain of '51':
XRCE_DOMAIN_ID_OVERRIDE=51 ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -r dds_xrce_profile.xml

This however didn't work, and it was still posting to domain 0. When the command is run it would state that the domain_id is overwritten to 51, as can be seen here:
[image]

We overcame this issue by adding '<domainId>51</domainId>' to the 'dds_xrce_profile.xml' file in 'AP_DDS'. With this line changed everything appears to work as expected, with it being posted to domain 51.

#### Expected behavior
Environmental variable should override setting of domain ID.

#### Actual behavior
dds_xrce_profile.xml overrides environmental variable

pablogs9, 2023-10-27:
It seems that this feature is only implemented for Binary entity creation and you are using Reference entity creation.

Let me open a quick PR for fixing this.

pablogs9, 2023-10-27:
Could you check that an Agent using this branch works as expected: https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/351 ?

pablogs9, 2023-10-30:
https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/351 merged and `ros2` branch forwarded to `develop`. This issue shall be solved. Please reopen if any problem.

closed 2023-10-30 by pablogs9