eProsima/Micro-XRCE-DDS-Agent #355: Cross Endian between Agent and Client

kimsos0723, 2023-11-17:
hi, I try to porting on PowerPC t2080rdb the XRCE-Client
I configured the Client to Big-endian and running `SubscribeHelloWorld` on ppc
 So result like-below 
[image]

but the Agent read  Xtype String length as Little-endian, even if submessage Header is 0x04

Did I do something wrong?

pablogs9, 2023-11-17:
Could you share a replicator?

kimsos0723, 2023-11-17:
I can't understand what replicator means.
but I just changed the Client's  and CDR's CMakeList `CONFIG_BIG_ENDIANNESS` options on

pablogs9, 2023-11-20:
Is correctly set the endianness flag on the ucdrBuffer structure:

void on_topic(
        uxrSession* session,
        uxrObjectId object_id,
        uint16_t request_id,
        uxrStreamId stream_id,
        struct ucdrBuffer* ub,
        uint16_t length,
        void* args)
{
    // Check this
    printf("ENDIANESS: %d\n", ub->endianness);

    (void) session; (void) object_id; (void) request_id; (void) stream_id; (void) length;

    HelloWorld topic;
    HelloWorld_deserialize_topic(ub, &topic);

    printf("Received topic: %s, id: %i\n", topic.message, topic.index);

    uint32_t* count_ptr = (uint32_t*) args;
    (*count_ptr)++;
}

kimsos0723, 2023-11-21:
Here's what the client sees as the received message
[image]

kimsos0723, 2023-11-21:
The length of a string of type Xtype seems to be misread as a little-endian in the agent.

pablogs9, 2023-11-21:
> Here's what the client sees as the received message [image]

Sorry, but I cannot retrieve enough information from this

kimsos0723, 2023-11-24:
Oh something changed, When I built the agent in the develop bench 
But previous error condition has passed but raised seg fault in Agent
[image]

closed 2023-11-28 by kimsos0723