eProsima/Micro-XRCE-DDS-Agent #357: Re-establishing connection to clinet on Jetson Nano

u19002409, 2023-12-01:
- Hardware description: Jetson Nano 4GB Devloper Kit
- OS: QEnignieering Ubuntu 20.04 Modidied IMage for Jetson Nano
- Installation type: MicroAgent communicating Via Serial at baud 230400 to pixhawk v1.14

#### Steps to reproduce the issue
In order to reproduce my error, I have to start the microagent and it occurs at random intervals.

#### Expected behavior
Establish a Connection and keeo that session.

#### Actual behavior
Creates a session and then restarts the session multiple times, here are examples of the session reastablishment, and my performance monitors
[image]
[image]
[image]
[image]
[image]
[image]

#### Additional information
I would like to know if anyone can help me with my particular issue. My suspicion is that the system creates a session either because of memory issues or CPU issues. I have monitored the system performance and upon monitoring I am not really convinced the tha CPU is the issue, at some times when CPU ussage is high the Agents will recreate a session, but sometimes it recreates a session without a high CPU usage, The memory usage is also within reason which leads me to think it must be either CPU or something else, any advise would be appreciated.

pablogs9, 2023-12-01:
Which is your Micro XRCE-DDS Client code?

u19002409, 2023-12-04:
It is the latest version from the github repository so I think its v2.4.2.

u19002409, 2023-12-04:
I do not believe I have one, or can you guide me where to get it.