eProsima/Micro-XRCE-DDS-Agent #358: Micro XRCE DDS Agent build without internet on armv8

15vanshika, 2024-01-31:
## Issue template

- Hardware description: Jetson Nano
- OS: Yocto with aarch64-linux 
- Installation type: CMake flags, FastDDS version, etc 
- Version or commit hash: version or branch used to compile the micro XRCE-DDS Agent

#### Steps to reproduce the issue
During build, the micro-XRCE DDS-agent attempts to pull content from github. But we dont have internet access , I want to know how can i build XRCE DDS agent without cloning I  tried to provide those repositories manually but it is  still failing .

pablogs9, 2024-01-31:
You can use CMake to build and install the agent dependencies (as you can see in [CMakeLists.txt](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/master/CMakeLists.txt): Fast CDR, Fast DDS and Micro XRCE-DDS Client). Once they are installed and CMAKE_PREFIX_PATH points to their installation, this repo shall be able to build setting [UAGENT_SUPERBUILD](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/57d086216d01ec43121845d385894a25987f8a2c/CMakeLists.txt#L23) to OFF, `UAGENT_USE_SYSTEM_FASTCDR` and `UAGENT_FAST_PROFILE` set to ON.

Probably you can skip the dependency with Micro XRCE-DDS Client turning OFF `UAGENT_P2P_PROFILE`

JaimeMartin, 2024-02-01:
Hi @15vanshika , 

This is not supported. As I mentioned yesterday in our call, these kinds of issues are addressed through commercial support.

closed 2024-02-01 by JaimeMartin