eProsima/Micro-XRCE-DDS-Agent #360: Build error for Linking CXX shared library libmicroxrcedds_agent.so

JustinHsuNctu, 2024-03-01:
Hi, I am trying to build micro XRCE-DDS Agent in Ubuntu 22.04. When I use "make" to compile the project, I get an error. Here is my bug report. How should I sove this issue?

[  2%] Linking CXX shared library libmicroxrcedds_agent.so
/usr/bin/ld: /usr/local/lib/libfmt.a(format.cc.o): warning: relocation against `stdout@@GLIBC_2.2.5' in read-only section `.text'
/usr/bin/ld: /usr/local/lib/libfmt.a(format.cc.o): relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/microxrcedds_agent.dir/build.make:842: libmicroxrcedds_agent.so.2.4.2] Error 1
make[1]: *** [CMakeFiles/Makefile2:85: CMakeFiles/microxrcedds_agent.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

pablogs9, 2024-03-01:
Could you provide a replicator or instructions you are following?

pablogs9, 2024-03-01:
Found no problem in Ubuntu 22.04: https://asciinema.org/a/ys5nUub96n24RuLUrQ0c0XRLr

JustinHsuNctu, 2024-03-04:
@pablogs9 Thank you for your quick reply. I think I found the problem! Before I installed DDS-Agent, I had already installed Pytorch, NVIDIA GPU driver, etc., for deep learning. They make a conflict with DDS-Agent, maybe.
I tried to reinstall Ubuntu, then install DDS-Agent. Now it is OKay!

closed 2024-03-04 by pablogs9