eProsima/Micro-XRCE-DDS-Agent #364: How to configure QOS settings on FastDDS entities created by clients

chrisg89, 2024-04-01:
Hello,

I understand the `micro-dds-agent` creates `FastDDS` entities to allow clients to produce and consume data in the DDS global data space (when `FastDDSMiddleware` is used).

How can the QOS settings of the FastDDS entities be configure? Specifically, we want to be able to select QOS properties on the domain participant such as:

- [TransportConfigQos ](https://fast-dds.docs.eprosima.com/en/latest/fastdds/python_api_reference/dds_pim/core/policy/transportconfigqos.html?highlight=user_transports#transportconfigqos)

On a related note, does the agent support IPC transport on domain participants facing the DDS global data space?

#### Steps to reproduce the issue
N/A

#### Expected behavior

#### Actual behavior

#### Additional information

pablogs9, 2024-04-02:
Hello @chrisg89, it is possible to create Fast DDS entities in the Micro XRCE-DDS Agent by reference, that means that you can load an XML defining your Fast DDS profiles and then use the Micro XRCE-DDS Client to instantiate this reference.

For example [here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/d44dc3fa0c488376e34d26ed92853f1c66dcb670/examples/ShapesDemo/main.c#L330) a Domain Participant is being created from the [this](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/master/agent.refs) definitions.

Here you have some more documentation: https://micro-xrce-dds.docs.eprosima.com/en/latest/client.html?highlight=references#creation-mode-client

chrisg89, 2024-04-05:
Thank you @pablogs9