eProsima/Micro-XRCE-DDS-Agent #365: Does an Agent using FastDDS have Discovery QoS Configuration?

Herbifors, 2024-04-13:
## Issue 

I tried applying the Discovery configuration from the link, but it doesn't seem to be working. Can an agent using FastDDS apply the QoS settings provided in the link when sending information via DDS?

#### Fast-DDS link
https://fast-dds.docs.eprosima.com/en/latest/fastdds/discovery/static.html#static-edp

pablogs9, 2024-04-15:
Yes, you can configure Fast DDS profiles using the environment variable `FASTRTPS_DEFAULT_PROFILES_FILE` as explained here: https://fast-dds.docs.eprosima.com/en/latest/fastdds/xml_configuration/xml_configuration.html

Herbifors, 2024-04-15:
> Yes, you can configure Fast DDS profiles using the environment variable `FASTRTPS_DEFAULT_PROFILES_FILE` as explained here: https://fast-dds.docs.eprosima.com/en/latest/fastdds/xml_configuration/xml_configuration.html

Thank you for your Answer. 
I've reviewed the document you provided and attempted to run a simple simulation. however, it seems that Static Discovery is not functioning in the Agent 

If Static Discovery is possible to the Agent, please let me know the method.

#### Simulate Task
1. Create an XML including &lt;EDP&gt;STATIC&lt;EDP&gt; and check the Discovery Change (Attachment 1)
   - FASTRTPS_DEFAULT_PROFILES_FILE="XML location"
   - run ros2 demos
   - 172.17.0.2 is apply the Static(fast-dds), 0.4 is normal Fast-dds
2. Apply the same XML file to the Agent and verify the Discovery Change (Attachment 2)
   - FASTRTPS_DEFAULT_PROFILES_FILE="XML location"
   - run "MicroXRCEAgent udp4 -p 8888"
   - 172.17.0.2 is apply the Static(agent), 0.4 is normal Fast-dds  and 0.2(agent) remained the same packet length regardless of the Discovery setting

Discovery changes were observed in FAST-DDS, but it seems that there were no changes in the Agent.

#### Attachment1
[image]
#### Attachment2
[image]