eProsima/Micro-XRCE-DDS-Agent #370: install error for previous versions related to 'fastdds'

kaesesahnetorte, 2024-07-19:
It seems like for some of the previous versions (<2.4.3) the installation broke.
See error message below.

#### Steps to reproduce the issue
install version 2.4.2

#### Actual behavior
Error message at step '$ make':
`
Cloning into 'fastdds'...
fatal: invalid reference: 2.12.x
CMake Error at .../Micro-XRCE-DDS-Agent/build/fastdds/tmp/fastdds-gitclone.cmake:40 (message):
  Failed to checkout tag: '2.12.x'
`

wntun, 2024-08-02:
for my case, I updated the line : checkout "2.12.x" to checkout "2.12.1".

leonhartyao, 2025-04-23:
I face the same issue with ros2 branch or tag v2.4.2.
Since it is built in a docker file, manually change the line in the generated file is not an option.
Could anybody point out where comes this tag?

Austin-Parks, 2025-06-16:
For me I had to change Line: 99 of CMakeLists.txt to the following:

set(_fastdds_tag v2.12.2)

Rck27, 2025-06-18:
changegd mine to 2.12.1 and 2.12.2 but both returns error of failed to checkout
 Failed to checkout tag: '2.12.2'

milesfrain, 2025-07-16:
fastdds "tag" `2.13.x` works, because _branch_ `origin/2.13.x` exists.

The problem is that there are many tagged releases of Micro-XRCE-DDS-Agent that now fail to build, because they depend on _branches_ (acting as "tags") of Fast-DDS that have been deleted last year. For example, Micro-XRCE-DDS-Agent `2.4.2` depends on the existence of Fast-DDS branch `origin/2.12.x`.
https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/57d086216d01ec43121845d385894a25987f8a2c/CMakeLists.txt#L99