eProsima/Micro-XRCE-DDS-Agent #372: Facing issue while running make on Ubuntu 22.04

yashrbhutada, 2024-07-31:
## Issue template
Facing multiple such issues

[image]
[image]
[image]

omarbo4k4, 2024-11-24:
the same

pablogs9, 2024-11-25:
Which OS are you using? Could you provide instructions for replicating the issue?

omarbo4k4, 2024-11-25:
I have solved the problem using a fresh installation of ubuntu 22.04 LTS 
So it needs to be very highlighted that in order to use this repo you need to have a brand new installation of ubuntu without running anything after installation but this repo 
Thanks. 

I have one more question for you if you can help with that 
How can i control the drone to move it in a target position, also is it possible to change the parameters of the model so i can run the same simulation on my own drone model (eg. The kv of the motors , the weight of the drone) 
Also is it possible to run my own pid control code or which folder does have the pid control so i can modify on it if it is needed to fit my own drone model 
Thanks in advance.

Tiqo01, 2024-12-13:
I’m encountering the same errors while trying to build in Yocto for the Raspberry Pi CM4. I disabled the UAGENT_LOGGER_PROFILE to get it to compile, but I would like to have the loggers enabled. 

yocto scarthgap
fmt : 10.2.1
spdlog : 1.13.0

Tiqo01, 2024-12-13:
I found where the problem comes from: when spdlog is built with -DSPDLOG_FMT_EXTERNAL=ON, I get this error.

KasperMollerHansen, 2025-02-10:
I'm experincing similar issues..

KasperMollerHansen, 2025-02-10:
I have managed to build it:

I deactivated my conda and ran the following lines: 

python3 -m pip install --upgrade pip
python3 -m pip install --upgrade setuptools wheel twine check-wheel-contents

Not sure if deactivating conda is necessary, i have not tried without

agntgalahad, 2025-05-31:
I also faced issues with installation.
Mine worked after I deactivated conda.