eProsima/Micro-XRCE-DDS-Agent #389: Agent crashes when receiving invalid Boolean values (not 0 or 1)

j4kb4dw0lf, 2025-04-28:
- Hardware description: x86_64 PC, standard desktop
- OS: Ubuntu 22.04 LTS
- Installation type: Built from source, CMake default flags, FastDDS 2.14
- Version or commit hash: commit hash `abcdefg12345` (main branch)

#### Steps to reproduce the issue
Send a packet that includes a boolean field with a value different from 0 or 1 (e.g., 2, 255).
You can easily modify a client to do this manually or fuzz the connection.

#### Expected behavior
The agent should either:
- Normalize the boolean value (e.g., treat anything nonzero as `true`), **or**
- Gracefully reject the packet and log an error.

#### Actual behavior
The agent crashes immediately when trying to deserialize a boolean field that has an invalid value (neither 0 nor 1). Exceptions are thrown by the FastCDR implementation of the deserializiation but are not caught, leading to undefined behavior and potential security risks (DoS by packet manipulation).

#### Additional information
This crash affects stability and opens the agent to simple denial-of-service attacks via malformed packets. A check for valid boolean values during deserialization would solve the problem, same would apply if the exception is gracefully caught.

4ntn, 2025-05-12:
Hello @j4kb4dw0lf ,

I am not being able to reproduce this issue.

Can you point me to somewhere in the code where you think this is happening? Or maybe specify how you are testing this.

On the other issue it was very helpful.

Thanks!

j4kb4dw0lf, 2025-05-12:
Hello @4ntn, in this casae the issue is very easily reproducible using the CREATE_CLIENT submessage which contains a boolean field: `m_properties_flag` that is deserialized here:
https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/155cfaaf8b7abac2e85d4a62d3649b09ace0be55/src/cpp/types/XRCETypes.cpp#L1130
if this value is found to be not 0 or 1, as by fastcdr policy:
https://github.com/eProsima/Fast-CDR/blob/36c28c4e5f638c1436c5e709d00ba9b68d358372/src/cpp/FastCdr.cpp#L380
it generates an exception which uncaught crashes the agent.

4ntn, 2025-05-13:
Hi @j4kb4dw0lf , thank you for the guidance.

I have implemented this fix: https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/393

I will close this issue when the PR is merged.

closed 2025-05-13 by 4ntn