eProsima/Micro-XRCE-DDS-Agent #390: MTU length not validated: possible call to alloc() with size 0

j4kb4dw0lf, 2025-04-28:
- Hardware description: x86_64 PC, standard desktop
- OS: Ubuntu 22.04 LTS
- Installation type: Built from source, CMake default flags, FastDDS 2.14
- Version or commit hash: 3.0.1 (main branch)

#### Steps to reproduce the issue
Force the MTU length parameter to be zero during client creation.
Trigger an event where an allocation for a message buffer is attempted, e.g. a response from the agent to a status message.

#### Expected behavior
The agent should detect invalid MTU values (e.g., <= 0) **before** calling memory allocation functions, and either reject the setup with a clear error or default to a minimal safe value.

#### Actual behavior
When the MTU length is not correctly checked, the alloc function can be called with a parameter of 0. This results in undefined behavior depending on the system's allocator and could lead to crashes, hard-to-debug issues, or memory corruption risks.

#### Additional information
Proper validation on the MTU field would prevent this behavior. Suggested fix: check that MTU is greater than 0 before calling any allocation functions and assert or log error otherwise.

j4kb4dw0lf, 2025-05-06:
Hello @4ntn , there is no way for me to refer to a specific line, if i were to tell you one in particular it would probably need to be here: https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/155cfaaf8b7abac2e85d4a62d3649b09ace0be55/include/uxr/agent/message/OutputMessage.hpp#L34 where the buffer for output messages are allocated. 

This is done in various instances of processing of submessages received by the agent (specifically in the push_output_submessage subsequent call).

The "setting" of the MTU lenght is actually part of a university project about software security where we were tasked to find potential security issues and bugs in the Agent, the campaign resulted in the two bugs i reported as issues.

The MTU lenght is specifically the one that is deserialized from a client_representation sent in a CREATE_CLIENT submessage (the last 2 bytes of said submessage).

In order to patch this issue and prevent crashes i would advise the approach of either setting a minimun mtu lenght value or adding addition checks to the session_info_.mtu used here: https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/155cfaaf8b7abac2e85d4a62d3649b09ace0be55/include/uxr/agent/client/session/Session.hpp#L272

4ntn, 2025-05-08:
Hello again, @j4kb4dw0lf ,

I have been able to reproduce the issue.
I have implemented this fix: https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/392

It would be nice if you could test it in your setup to confirm.